bitraf/bitraf-cnc #3: Stepper drivers

Jaknil, 2018-10-11:
What does the pins on the driver do? Check these:
* GND 
* +5V (Does it provide or need 5V here?)
* DIR (Most likely: 5V HIGH/LOW  sets direction)
* STEP (Most likely: short 5V HIGH pulses increments the stepper motor a step/microstep)
* CLEAR (Possibly: Clear alarm by 5V HIGH or LOW?)
* X ( RES) (?)
* ALARM (Assume: Goes 5V HIGH or LOW when the stepper can't reach position, which one?)

What is the pin spacing on the card? (To be able to design a connector)

What does the second pin do for each pin? Is it to be able to pull it HIGH/LOW permanently?

How many steps per mm does the shopbot take? Is it compatible with the standard GRBL pulse length?

Bonus: Is there any settings we want to tweak on the stepper driver like closed loop error tolerance?

Jaknil, 2018-10-16:
I ran the numbers and the setting for 1000 steps / revolution should work fine. We will get a max speed of 542mm/s (32,5m/min) at 30kHz. The smallest movement will then be 0,02mm (/ step) which isn't fantastic but should absolutely be good enough. (Calculated for 130mm pinon).

What I couldn't calculate my way to is the value of  97.748705 steps per mm. Were did that come from? Is it a shopbot setting? Odd if so. 

Value | Unit | Comment
-- | -- | --
1000 | puls/rot pre gearbox | Stepper driver setting
7,2 | gearbox | ShopBot PRSalpha
7200 | steps per rot, |  
0,05 | deg per pulse ink gearbox |  
130 | cicumference ca? |  
0,018 | mm/pulse | Min movement, ok
55,38 | steps/mm with these settings |  
97,748705 | steps per mm | Shopbot vaule from jens?
30000 | puls per sec max from arduinno |  
4,17 | rotations per sec max |  
541,67 | mm/s | Max travel speed, more than needed
32,5 | m/min |  
45,7 | m/min | shopbot stock max speed, never used

JensDyvik, 2018-10-16:
I am puzzled as well by the default values in the shopbot firmware.

The shopbot drivers are capable of microptepping x5 and x10. They can also do half steps. I am not sure what stepping level shopbot3 uses. Default shopbot3 settings do use different step levels for jog and and cut commands. If our current setting is 500steps *(before gearing)* for jogs and 1000 *(before gearing)* during cuts, we are all good with the GRBL firmware.

If what we are accustomed to today is 5,000 or 10,000 steps *(before gearing)* during cuts, we need to do some testing to verify that the reduction in motion smoothness and precision is acceptable.