bitraf/bitraf-cnc #4: Spindle control

Jaknil, 2018-10-11:
How to interface our new controller with the spindle on the shopbot?

The situation:
HRBL controls one optocoupler transistor to turn on the spindle in the forward direction and another to create a 1kHz PWM for speed control. 

The shopbot Variable Frequency Drive (VFD) drives the spindle. The default shopbot way of controlling it is via a 2 pin MEMOBUS/Modbus comm. RS-485/422 serial communication. [ Here is the shopbot manual for connecting it](http://www.shopbottools.com/ShopBotDocs/files/RPMControllerYaskawa%202015%2004%2006.pdf)

And here is the [VFD sellers product page](https://www.yaskawa.com/products/drives/industrial-ac-drives/microdrives/v1000-drive)

Alternatives:
1.  We can use another microcontroller to interpret between the PWM signal and the serial communication.
2. We can use analog control of the VFD like we do in fabricatable machines. There is an option on the VFD to control it with a reference voltage between 0-10V and they provide a 10,5V source

Nr 2 Sounds super easy, no new components required, we just need to route the traces differently.

If you are OK with nr2 then I can implement it on the card.

tingox, 2018-10-11:
Either way, we should test that it works the way we want before committing to a solution. Solution nr 2 might work differently from the way it works today (different range, responsiveness, or other) in a way that makes us unhappy. If it works exactly like today we're fine.

JensDyvik, 2018-10-11:
I agree that we should test as soon as possible. It is easy to open the VFD case and connect new wires. It might be that the VFD needs reprogramming to listen to pwm input instead of RS-485. In that case we need to be sure that we can program it back easily too to avoid a lengthy downtime after testing.

@Jaknil can you take responsibility for testing pwm from an arduino to the VFD?
@tingox if jakobs test fails, can you write an arduino sketch for converting pwm to RS-485 and source the serial interface hardware(shield)?

Note that the shopbot v1000 VFD is a special IP66 type and might have differnt programming/interface. I have not found the v1000 ip66 manual online yet

Jaknil, 2018-10-12:
Yes I can be responsible for testing analog PMW control on the bitraf shopbot. I assume the fellesverkstedet ones are not easily accessed for testing? It does need repogramming and dip switch changes. I will read up properly before and 

I have simulated some PWM smoothing circuits now using [this site](http://sim.okawa-denshi.jp/en/PWMtool.php) and think R 1000 ohm and C 10uF are good values for the PWM smoothing. It leaves a 2,5% noise in RPM but has a super fast response for future cornering RPM control. It can do 0-100% in 0.05 s. 
[image]

JensDyvik, 2018-10-12:
Super. Yes, definitely better to test VFD control by PWM at bitraf.

2.5% noise with super fast response rate sounds great. Without any filter we only have very minor fluctuations in spindle speed on Hattori at bitraf.

From: Jakob Nilsson 
Sent: Friday, October 12, 2018 11:56 AM
To: bitraf/bitraf-cnc 
Cc: Jens Dyvik ; Comment 
Subject: Re: [bitraf/bitraf-cnc] Spindle control (#4)

Yes I can be responsible for testing analog PMW control on the bitraf shopbot. I assume the fellesverkstedet ones are not easily accessed for testing? It does need repogramming and dip switch changes. I will read up properly before and

I have simulated some PWM smoothing circuits now using this site and think R 1000 ohm and C 10uF are good values for the PWM smoothing. It leaves a 2,5% noise in RPM but has a super fast response for future cornering RPM control. It can do 0-100% in 0.05 s.

Jaknil, 2018-10-12:
@tingox offered to build the test circuit. I think this is what we need:
I have optocouplers if there are no nice transistors kicking around in the lab.

Let's mark the wires as well!

[image]

I uploaded the Kicad folder to the [repo](https://github.com/bitraf/bitraf-cnc/tree/master/Hrbl_for_shopbot/pwm_smooth)

Jaknil, 2018-10-12:
Reminder to self: Need to double check the R and C values for 10V

JensDyvik, 2018-10-12:
Awesomely fast work! Looking forward to hearing the results.

I have started work on super simple test board for testing to controll only one driver. It has only one male 2x8 pin connector for the ribbon cable to the driver and female connectors for an arduino nano.

tingox, 2018-10-15:
[image]
Here is the test circuit (without wires) @cisc and I built it, and he verified the wiring
Parts substituted:
2N3904 transistors
10 mikroF 25V electrolytic capacitor
I have checked the datasheet, and 2N3904 should work for the voltages and currents we will use. I'm not sure if the electrolytic capacitor is a good enough substitute, @jaknil ?

tingox, 2018-10-18:
Questions we need to figure out before the test:
- security circuit: do we need to bypass it (to run with the door to the electronics open for example)?
-- and if needed, how?
- mains power to the VFD: just rig a signal that activates the contactor? Or do we need to do it another way?

Jaknil, 2018-10-19:
The spindle test was't successful but it wasn't a failure either. We couldn't set all the parameters for some reason and we accidentally used an old 0.9 version GRBL which used a different pinout from what we wired it for. We learned a bunch though and could restore it to working order inside our booking.

Here is the full test instructions and test log: https://docs.google.com/document/d/1vSm9jWW1w2j-wPlAy6cMJmsO1I4x_Cvt3CuZz1CbSg8/edit?usp=sharing

[image]
[image]