bitraf/bitraf-cnc #7: Requirement: auxilliary signal from contactors shall activate abort signal

tingox, 2018-10-11:
The shopbot has contactors that disable (cut) power to stepper motor drivers and VFD when alarm or emergency stop signal happens. These contactors have an auxillary signal (24V) that is used to detect if the contactor works. 

This auxillary signal shall be used to activate the abort-signal on the new controller if one of the contactors fail.

JensDyvik, 2018-10-11:
as discussed in this [issue](https://github.com/bitraf/bitraf-cnc/issues/6#issuecomment-429093533) we have in the existing hrbl card a detection of 5V loss that sets the abort high. but currently the contactors does not kill power 5V power supply. since the contactors already have a 24V guidance (aux) signal it should be super easy to listen to this instead of 5V power loss

Jaknil, 2018-10-12:
Yes that can easily be implemented. Also with a information LED like discussed in Issue #8 

"but currently the contactors does not kill power 5V power supply" - Doesn't that depend on how we power our card? The on-card 5V regulator (needed for the optocouplers) can be powered by the stepper driver PSU like we did on Hedy? The 5V from USB is separate.

JensDyvik, 2018-10-12:
The shopbot does not have a PSU for the drivers. The drivers run directly on 220V single phase power.

The shopbot cabinet has a 24v and a 5v power supply. The safety contactor does not kill their power.

It might be unecceary complex to set up an extra relay/mosfet to kill power input to our card, when all we need is to listen to the 24V aux signal from the safety contactors.