bitraf/bitraf-cnc #8: errors: better feedback

tingox, 2018-10-11:
We want better feedback about errors: is the cause of an error an alarm from a stepper motor driver, from emergency stop (estop), or something else?

tingox, 2018-10-11:
The final version should report errors via the user interface / control program. for a first version a display or some LEDs would be acceptable.

JensDyvik, 2018-10-11:
It would also be great to report which axis has created the alarm. For instance an alarm message like "X axis is not able to move. Check for obstructions and restart" would be very amateur friendly

Jaknil, 2018-10-12:
In this first version we can implement LEDs on the front of the card with laser-silkscreen information about what caused the error. 

Sources of error which can be shown with LEDs:
* One for each axis motor(s) driver in ALARM state
* External power loss
* Safety contactor not connected

The arduino already has leds that indicate successfull USB connection (the power led) serial comminication (the RX TX LEDS) and spindle RUN signal (pin 13 built in led).

The Universal Gcode sender interface can already detect and show if we are in a general alarm state, which limit switch or if the probe is triggered. So I don't think we need leds for these.

JensDyvik, 2018-10-12:
I vote that we dont implement any of this in the first version of the card. Our task list is already quite long. Aslo, the oriental motors drivers already have nice blinking leds to indicate state