bitraf/bitraf-cnc #10: Z probe: some tools (some milling bits) are not electrically conductive

tingox, 2018-10-12:
The current idea for a Z probe is to use a metallic plate off the work area somewhere to measure correct Z height / zero by measuring electrical conductivity between the plate and the tool.
However, some tools (example: diamond milling bits) are not electrically conductive. Shopbot sells a pressure sensitive Z-Zero plate https://store.shopbottools.com/products/pressure-sensitive-z-zero-plate
that can be used in this situation.
Should we select electricallly conductive or pressure sensitive technology?
Or can we have both?

JensDyvik, 2018-10-12:
I vote that buy and implement the shopbot pressure probe. It would be nice to experience pros and cons with this one compared to a conductive aluminium plate.

We already have good experience with self made alu probe plates, and can easily revert to this if are not happy with the pressure sensitive probe

Jaknil, 2018-10-15:
Found the manual for it: http://www.shopbottools.com/ShopBotDocs/files/PressureSensitiveZzero.pdf