bitraf/bitraf-cnc #13: Help needed for milling single axis test board

JensDyvik, 2018-10-15:
We have the production DXF ready for a single axis board [here](https://github.com/bitraf/bitraf-cnc/blob/master/Hrbl_for_shopbot/single-axis-test-CNC-ready.dxf). Kicad source is also in the repo.

Does @tingox or @Jaknil have the opportunity to mill this at bitraf?

The board is single sided and designed for 0.4mm or smaller milling bits. PCB stock can be found in bitmart.

The goal with this board is to:
- verify that we can control the oriental motors drivers with grbl,
- check if we have understood the pin connections correctly
- check if the 30000HZ limit of grbl will give us sufficient top speed

tingox, 2018-10-16:
there are free slots (in the booking calendar) this week, but I have no experience milling PCB's, so I would need help from someone with experience.

CISC, 2018-10-19:
Using random bits lying around was probably not a recipe for success (nor was using a rather well used spoil board). :)

We have a specific tool for this in Bitmart BTW, the Isolation Engraving bit which combines a tiny V-bit with a short straight flute.